StephenCarlson/MiniHawk-VTOL #22: Branding and Logo Design

StephenCarlson, 2021-01-19:
This project needs a good logo. The following guidelines are suggested:

- It probably isn't a good idea to use the aircraft outline or shadow/form as part of the icon. As the aircraft design shape changes, this will not age well. 
- Perhaps a bird outline will do, but something that evokes "VTOL" in the imagery. "Mini" and "Hawk".
- The entire text should be only the following "MiniHawk-VTOL", with the option for VTOL being separate and below and stylized differently from the rest. "MiniHawk-VTOL" is one line, no spaces, and CamelCase capitalization for the 'M' and 'H' in the name.
- The logo should be able to be reproduced using only written instructions, and mildly accessible to anyone with a copy of Inkscape? Anything too complicated or specific to a particular image editing suite will make it difficult to maintain or reproduce in the future.
- Obviously, the logo should be in vector format, and rendered at various sizes.
- The precursor to the MiniHawk was the OrangeHawk, which had a simple text logo (Orange Cursive Font with black outline) as seen in this image here: https://diydrones.com/profiles/blogs/orange-hawk-vtol-update-december-2013 . The Cursive font is lost to my current laptop file collection, I'll have to dig to find the actual font used. It might be best to use this font and simply repeat in the same manner.
- Making a social media square logo for the project necessitates that there be an actual logo design, so simply doing the name in cursive font isn't going to work.....

Here are some excellent references from the UAV world to imitate:
https://px4.io/brand-guidelines/
https://ardupilot.org/dev/docs/trademark.html
https://commons.wikimedia.org/wiki/File:SpaceX-Logo.svg
https://github.com/betaflight/betaflight/blob/master/docs/assets/images/bf_logo.png

StephenCarlson, 2021-01-28:
Possible Fonts for the logo:
https://fontsgeek.com/fonts/Megahertz-In-Regular
https://fontsgeek.com/freestyle-script-font

nermish, 2021-09-04:
while a graphic designer I am not, your ask for a bird with VTOL themes immediately made me think of a Kestrel. They're known for their odd hunting practice of hovering at altitude with remarkable stability, and have a very recognizable silhouette. I threw this together in the last hour or so, feel free to use it if you like it, or if not, maybe it'll spark an idea for someone else later on!

[image]
[MiniHawkVtol-SVG.zip](https://github.com/StephenCarlson/MiniHawk-VTOL/files/7109307/MiniHawkVtol-SVG.zip)

hope this project isn't dead, looks very promising!

StephenCarlson, 2021-09-04:
@nermish  Thank you so much for the logo, this helps! The project is definitely not dead, just in a very weak stealth mode :) . Per the Hackaday.io post I made in April, I've been wanting to make some grand release with an awesome music video to boot, but being saturated with research all summer, I've been unable to really focus on it that. (Connected to that is updating the readme and parts list.) But the good news; all my summer research has been directly pertaining to this vehicle and some interesting things that you can do with multiple MiniHawks flying at the same time. I've built at least three units of the V2 design as released in April, and have flown each extensively. 

My thoughts on the logo: 
- I like the Kestrel silhouette, but it will drive me and any other ornithologically-minded people crazy that it isn't a hawk .... if that can be deduced from the silhouette? It's great otherwise! BTW, how did you capture the outline?
- Does the octagon have any driven ratio or driven geometry? As in, what is the curve or tension on the splines that make up the outside perimeter? Reverse-engineering this in Inkscape, it looks like it is made using the polygon tool with Rounded=0.060?
- The hue is a pure share of orange a 24 degrees on the color wheel; mind if that is shifted to 32 or 36 degrees? I guess the text can stay at 26 degrees, or does that look best also shifted to 32/36?
- Imitating the PX4 project, where there are multiple versions of the logo with dark and light versions, below is my modification for a light-version, with sharper corners for the octagon on the outside edge (keeping soft inside corners), and changing the outside color to 32 degrees while retaining 26 degrees for the text:
[image]

Let me know what you think, lets keep iterating on this, and again, thanks for putting this up!

StephenCarlson, 2021-09-04:
@nermish I also tried a seven-corner outline, if this looks more "cool" and balanced? It has a minor relationship to the planform, which if connecting corners, becomes a seven-corner polygon as well:
[image]

StephenCarlson, 2021-09-04:
[image]

nermish, 2021-09-04:
Those revisions look incredible! I agree that the seven pointed outline looks much better, and the rounded internal corners give some good visual depth. 

> Regarding bird choice, I'd always heard them called "Sparrow Hawks," and didn't realize they weren't actually hawks. Their outline is decently unique, so if you'd like to try a few others out, to make the silhouette I first  found an image of a hovering one online and loaded it into paint.net (any image editor would work) and used the magic wand selection to select the outline, then filled the whole bird in with a solid color. I took that image to an [online vectorizer](https://www.pngtosvg.com/)  and set the smoothing to a low value so the rough edges disappear. After that i just downloaded the SVG and put that into inkscape!
 

> I didn't really have any driven dimensions for the polygon, though i probably should have. it was indeed made in inkscape and eyeballed with the polygon tool 😁

>I think the 36 degree orange looks better, it's a bit more vibrant and that's good
 
>I won't be back at my PC until later today, but I'd also tried a textless version with just the bird that could likely make a good social media logo or graphical signature on documentation, but I couldn't make it look right with the octagon. On the septagonal background, it looks like it'd be much more feasible! 
 
I'm glad I could help with this, feel free to modify that logo until you find something you like, it's yours as far as I'm concerned

StephenCarlson, 2021-09-19:
@nermish I've been working on the logo design some more, but I just realized while looking at my backpack, the grey hawk encircled with an orange/black circle is pretty much the same as the Osprey Backpack company logo. Ugh! Or is that fine?