StephenCarlson/MiniHawk-VTOL #47: Enhancement camera embedded in  nosecone

davidabevan, 2022-11-25:
Thanks for great work so far.
Am attempting this as winter project. My 2nd ardupilot vtol. I am wondering if a new nosecone can incorporate the fpv camera in the front or such like. Yes the possibility of camera damage is there but lots of fpv planes do use this approach. Hopefully you can think on this maybe adapt your design later. Thanks for sharing project.

StephenCarlson, 2022-11-26:
I like leaving the nose tip empty for use as a ballast location. (I use between 10g and 40g of lead weight in the nose tip for my MiniHawks.) The FPV Hatch/Lid STL can support a 15mm cube camera and VTX just fine while keeping it out of the dirt and damage. But, yeah, I agree that the placement is somewhat annoying if you want to see the bottom-arch of the video feed that is obscured by the nose for that position.

davidabevan, 2022-11-26:
Thanks for getting back to me.. Ok no probs.
Ive also read your comments on the "Jumper Fiasco" I totally agree with you and think they are completely in the wrong. It also appears they made a Lemon due to substandard electronics no dshot esc and FC reboots. NO Compass  WHAT>>>>Just think what COULD have happened with some collaboration between yourself and the ardupilot community.
My Thanks and support Dave B

davidabevan, 2022-12-07:
Thanks for Reply Sorry I forgot to close...

closed 2022-12-07 by davidabevan