StephenCarlson/MiniHawk-VTOL #48: CAD files

madhunm, 2022-12-27:
Stephen,

Do you think it would be possible to release the CAD files (STEP format) for the latest revision of MiniHawk?

Thanks!

StephenCarlson, 2023-01-19:
@madhunm That.... is a very relevant question, and one I'm grappling with. I released the Hatch here on GitHub in STEP format, in the hopes that it would allow others to make custom hatch mods, but I held off from giving all the STLs the STEP treatment since it would make reverse-engineering the design for mass production a breeze, and I figured that I would re-evaluate that decision if the hatch STEP mods idea got popular. Well, the drama of the past few months with Jumper reusing the MiniHawk for an unsanctioned and unattributed commercial implementation has shown that releasing the majority of the aircraft in STL format only ( which is relatively low-res compared to STEP) didn't stop them from running with it anyway. (Also note that I released the Nacelle design as a STEP assembly to the Facebook VTOL Hangout Group, I had hoped that the more tight-knit community there might be a safer place to post that, but it may well have also been scraped by Jumper.) 

A few ideas for what I can do:
1. Keep the pattern that currently exists: Only STLs except a few select STEP files.
2. Provide the STEP version to individuals on a case-by-case basis
3. Release everything as STEP.

I'm leaning toward the second option. What would you like to do with the STEP version?

-Steve

jhiggason, 2023-01-27:
Do you have a post talking about the Jumper drama?  It does look like they completely ripped off your design...I kind of liked some of the jumper products but will stop buying them if they are straight up just ripping off others work.  (I'm trying to find a good, low cost radio to reccomend to my "customers" and will definately not be using jumper if this is their typical business practice)

StephenCarlson, 2023-01-27:
Yep, it is discussed here, just one tab over in "Discussions". https://github.com/StephenCarlson/MiniHawk-VTOL/discussions/46

EreminAleksey, 2023-11-06:
Hi Stephen.
Could you please provide the minihawk step files. I want create and make 3d prinded molds for polyurethane foam. I don`t want to buy xiake800 body, because jumper stole your project.

mohamed12354, 2025-12-02:
hi steve 
can we see the draw dimension of drone ?